makerdiary/nrf52840-mdk-usb-dongle #116: Problem with Setting up the environment guide

ivan-ushakov, 2025-06-11:
I use Docker and Dev Container with VS Code and followed this tutorial about setting up environment - https://wiki.makerdiary.com/nrf52840-mdk-usb-dongle/guides/ncs/setup. I did all steps without errors but now I can not build example project:

```bash
vscode ➜ /workspaces/engine-tester-device/my-workspace/ncs-playground (main) $ west build -p always -b dongle_nrf52840 samples/blinky
-- west build: making build dir /workspaces/engine-tester-device/my-workspace/ncs-playground/build pristine
-- west build: generating a build system
Loading Zephyr module(s) (Zephyr base): sysbuild_default
-- Found Python3: /usr/bin/python3 (found suitable version "3.10.12", minimum required is "3.10") found components: Interpreter 
-- Cache files will be written to: /home/vscode/.cache/zephyr
-- Found west (found suitable version "1.4.0", minimum required is "0.14.0")
-- Board: dongle_nrf52840
Parsing /workspaces/engine-tester-device/my-workspace/zephyr/share/sysbuild/Kconfig
/workspaces/engine-tester-device/my-workspace/zephyr/scripts/kconfig/kconfig.py: soc/Kconfig.v2:39: '/workspaces/engine-tester-device/my-workspace/ncs-playground/build/Kconfig/soc/Kconfig.soc' not found (in 'source "$(KCONFIG_BINARY_DIR)/soc/Kconfig.soc"'). Check that environment variables are set correctly (e.g. $srctree, which is set to '/workspaces/engine-tester-device/my-workspace/zephyr'). Also note that unset environment variables expand to the empty string.
CMake Error at /workspaces/engine-tester-device/my-workspace/zephyr/cmake/modules/kconfig.cmake:365 (message):
  command failed with return code: 1
Call Stack (most recent call first):
  cmake/modules/sysbuild_kconfig.cmake:116 (include)
  cmake/modules/sysbuild_default.cmake:19 (include)
  /workspaces/engine-tester-device/my-workspace/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:75 (include)
  /workspaces/engine-tester-device/my-workspace/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:92 (include_boilerplate)
  /workspaces/engine-tester-device/my-workspace/zephyr/share/sysbuild-package/cmake/SysbuildConfig.cmake:8 (include)
  template/CMakeLists.txt:10 (find_package)

-- Configuring incomplete, errors occurred!
See also "/workspaces/engine-tester-device/my-workspace/ncs-playground/build/CMakeFiles/CMakeOutput.log".
FATAL ERROR: command exited with status 1: /usr/bin/cmake -DWEST_PYTHON=/usr/bin/python3 -B/workspaces/engine-tester-device/my-workspace/ncs-playground/build -GNinja -DBOARD=dongle_nrf52840 -S/workspaces/engine-tester-device/my-workspace/zephyr/share/sysbuild -DAPP_DIR:PATH=/workspaces/engine-tester-device/my-workspace/ncs-playground/samples/blinky
```

Something wrong with my environment?

ivan-ushakov, 2025-06-11:
Solved with this comment (https://github.com/makerdiary/ncs-playground/issues/3#issuecomment-2109393574):

> in the ncs-playground/west.yaml file change the sdk-nrf revision from main to v2.5.3 and run west update.

cyang812, 2025-08-12:
Hi, after I change to v2.5.3 and rebuild, I encounter below error. 

```
-- west build: making build dir D:\github\my-workspace\ncs-playground\build pristine
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: D:/github/my-workspace/ncs-playground/samples/blinky
-- CMake version: 4.1.0
-- Using NCS Toolchain 2.8.20241119.782234748775 for building. (C:/ncs/toolchains/b620d30767/cmake)
CMake Deprecation Warning at D:/github/my-workspace/zephyr/cmake/modules/FindDeprecated.cmake:121 (message):
  'PYTHON_PREFER' variable is deprecated.  Please use Python3_EXECUTABLE
  instead.
Call Stack (most recent call first):
  D:/github/my-workspace/zephyr/cmake/modules/python.cmake:16 (find_package)
  D:/github/my-workspace/zephyr/cmake/modules/zephyr_default.cmake:129 (include)
  D:/github/my-workspace/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:66 (include)
  D:/github/my-workspace/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:92 (include_boilerplate)
  CMakeLists.txt:8 (find_package)

-- Found Python3: C:/ncs/toolchains/b620d30767/opt/bin/python.exe (found suitable version "3.12.4", minimum required is "3.8") found components: Interpreter
-- Cache files will be written to: D:/github/my-workspace/zephyr/.cache
-- Zephyr version: 3.4.99 (D:/github/my-workspace/zephyr)
-- Found west (found suitable version "1.2.0", minimum required is "0.14.0")
CMake Error at D:/github/my-workspace/zephyr/cmake/modules/zephyr_module.cmake:77 (message):
  ERROR: project sdk-nrf (nrf): cannot import contents of west.yml; do you
  need to run "west update"?

Call Stack (most recent call first):
  D:/github/my-workspace/zephyr/cmake/modules/zephyr_default.cmake:129 (include)
  D:/github/my-workspace/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:66 (include)
  D:/github/my-workspace/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:92 (include_boilerplate)
  CMakeLists.txt:8 (find_package)

-- Configuring incomplete, errors occurred!
FATAL ERROR: command exited with status 1: 'C:\Program Files\CMake\bin\cmake.EXE' -DWEST_PYTHON=C:/Python313/python.exe '-BD:\github\my-workspace\ncs-playground\build' -GNinja -DBOARD=dongle_nrf52840 '-SD:\github\my-workspace\ncs-playground\samples\blinky'
```

ivan-ushakov, 2025-08-12:
Hi @cyang812, have you ran `west update`? What was the output?